muskie82/MonoGS #128: How can this be run on a linux server without gui?

derrick-xwp, 2024-08-10:


ShreyaIAB, 2024-08-12:
I was able to launch GUI by  typing this command - **export MESA_GL_VERSION_OVERRIDE=4.5** before launching the python script

UltraHertzz, 2024-08-14:
just set use_gui to false in config. or add  --eval args in the console, it will run in headless mode, but as a drawback, you can not see any visual information during runing.

scpoppin1207, 2024-10-18:
> just set use_gui to false in config. or add --eval args in the console, it will run in headless mode, but as a drawback, you can not see any visual information during runing.

But it still shows "ModuleNotFoundError: No module named 'tkinter'". Could you please detail your method to run without GUI?

xxje-library, 2025-04-10:
@scpoppin1207  I use the software "MobaXterm", I have seen the GUI and it is running normally

scpoppin1207, 2025-04-10:
Thanks. I will try it latter. 😃

Cornell-CUG, 2025-08-12:
> [@scpoppin1207](https://github.com/scpoppin1207) I use the software "MobaXterm", I have seen the GUI and it is running normally
i use the MobaXterm too，but the screen has been stuck at the initial interface without refreshing