muskie82/MonoGS #140: Processing of custom datasets

MikeAiJF, 2024-09-04:
I need to use my own dataset for inference now, but the only difference between my dataset and the author's dataset is the format of the images. What method should I use to obtain the same format as the dataset provided by the author

hnglp, 2024-09-04:
Can you explain how you created the groundtruth information in the dataset structure for your own custom dataset? Or is this part what you mean by format difference?

MikeAiJF, 2024-09-04:
> 您能解释一下您是如何为自己的自定义数据集在数据集结构中创建真实信息吗？或者这部分就是您所说的格式差异？

[image]
How can I obtain the data in the picture

MikeAiJF, 2024-09-04:
My dataset only contains images from the RGB folder

MikeAiJF, 2024-09-04:
How should I obtain other data

hnglp, 2024-09-04:
This is the tricky part anyway. I am trying to proceed with rgb and groundtruth data without depth information. There are some exif tools for this, but I haven't gotten the result I want yet. If I make progress, I will share it here.

MikeAiJF, 2024-09-04:
That's OK. Thank you very much!

ShreyaIAB, 2024-09-04:
Hi, i am also trying with RGB images only. please share