muskie82/MonoGS #150: Significant Errors in Trajectory and Reconstruction across Various Datasets

hnglp, 2024-10-18:
I am trying to test the system in outdoor environments using different datasets and configurations. Below is a detailed breakdown of the methods I used:

- KITTI Odometry (Processed using TUM-like parser and TUM-style groundtruth data)
- KITTI Odometry (TUM-style groundtruth data processed using the TUM parser, combined with LiDAR-based depth maps)
- KITTI Odometry (Original KITTI data format processed with pykitti tool and my custom parser, without depth images)
- KITTI Odometry (Same approach as above, but depth maps were generated from projected Velodyne point clouds)
- KITTI Raw Dataset (Two different subsets processed using pykitti, no depth images included)
- KITTI Depth Dataset (Processed using only RGB images and my custom parser)
- NCLT Dataset (Encountered an error indicating "insufficient motion"; system could not be run successfully, retry needed)

For each method listed above, the system was run at least once. However, I consistently encountered significant errors in both trajectory estimation and reconstruction across all setups.
Possible Causes of the Errors:

- Groundtruth Issues: I tried both the original KITTI groundtruth format and the TUM-style version. Despite this, groundtruth processing may still be causing errors.

- Optimization Parameters: The optimization settings were tuned for KITTI's original data, and additional improvements were applied. However, the system appears to work better with indoor datasets, suggesting that factors like sunlight or outdoor conditions might be contributing to the issues.

- Frame Processing Delays: There is a notable difference in the frame processing times between TUM and KITTI datasets. For instance, processing 20 frames takes 0.6 seconds for TUM, whereas for KITTI, the same amount of frames takes 1.9 seconds.This delay might be leading to skipped frames and insufficient time for accurate reconstruction.

If anyone has any ideas or suggestions to help resolve these issues, your input would be greatly appreciated!

joontack-han, 2025-02-18:
Hi are you still working with it? or have you solve the problem?

hnglp, 2025-02-27:
I have stopped working with outdoor datasets for now. It seems that the system performs well when the camera moves around a static object, but it struggles in scenarios where the camera is fixed, and the environment moves. This issue might be resolved with loop closure, and it is also crucial to be careful while preparing ground truth data. I found a dataset—though I can't recall its name now—that I manipulated to achieve slower camera movements, and I obtained reasonably good results. However, due to my system's limitations, I couldn't complete a full session as my RAM kept running out. In summary, I have put outdoor studies on hold for now.

joontack-han, 2025-03-02:
Thanks for your kind reply. So you are saying that you found dataset that works pretty well with MonoGS right?