muskie82/MonoGS #157: question about camera pose optimization

Torment123, 2024-11-13:
Hi, I have a question about the initialization of camera poses, in line 121 of slam_frontend.py, it shows that the poses are initialized with the ground truth poses extracted from the dataset, I'm a bit confused here, because based on my understanding the goal of camera optimization is to the ground truth pose of the views through reconstruction based on equation 7 of the paper. If that's the case, why is it appropriate to initialize the viewpoints with ground truth poses? This might be my mistake here since I'm new to SLAM.

victkk, 2024-11-15:
> Hi, I have a question about the initialization of camera poses, in line 121 of slam_frontend.py, it shows that the poses are initialized with the ground truth poses extracted from the dataset, I'm a [...]

The function is only called for the first frame or when the SLAM system lose track completely(which will then start a new map). It's basically just setting the coordinates.

Torment123, 2024-11-15:
> The function is only called for the first frame or when the SLAM system lose track completely(which will then start a new map). It's basically just setting the coordinates.

Got it, thanks for replying

closed 2024-11-15 by Torment123