muskie82/MonoGS #163: AttributeError: module 'evo.core.trajectory' has no attribute 'align_trajectory'

Emmanuel9012, 2024-11-29:
I am using conda env (python=3.7.12) to impliment MonoGS SLAM. I face this error and cant overcome it. When i import evo on terminal there is no error. But on VS code there is this error "Import "evo" could not be resolvedPylancereportMissingImports" there python interpreter selected right.

python slam.py --config /mnt/data/Students/Aman/MonoGS/configs/mono/tum/fr1_desk.yaml
MonoGS: saving results in results/data_TUM_RGBD/2024-11-28-22-44-28
FEngine (64 bits) created at 0x555c48c43a20 (threading is enabled)
FEngine resolved backend: OpenGL
Failed to open X display. (exiting).
MonoGS: Resetting the system
MonoGS: Initialized map
Traceback (most recent call last):
File "slam.py", line 252, in
slam = SLAM(config, save_dir=save_dir)
File "slam.py", line 110, in init
self.frontend.run()
File "/mnt/data/Students/Aman/MonoGS/utils/slam_frontend.py", line 355, in run
monocular=self.monocular,
File "/mnt/data/Students/Aman/MonoGS/utils/eval_utils.py", line 110, in eval_ate
monocular=monocular,
File "/mnt/data/Students/Aman/MonoGS/utils/eval_utils.py", line 29, in evaluate_evo
traj_est_aligned = trajectory.align_trajectory(
AttributeError: module 'evo.core.trajectory' has no attribute 'align_trajectory'
MonoGS: Resetting the opacity of non-visible Gaussians
^CSIGINT: Interrupted by Ctrl-C.
SIGINT: Interrupted by Ctrl-C.

ShaySheng, 2024-12-22:
Same issue, no idea what's happening...

(MonoGS) root@autodl-container-3fac46b1d9-046c9e43:~/MonoGS# python slam.py --config configs/rgbd/tum/fr1_desk.yaml
MonoGS: saving results in results/datasets_tum/2024-12-22-23-37-16
FEngine (64 bits) created at 0x548968c0 (threading is enabled)
FEngine resolved backend: OpenGL
Failed to open X display. (exiting).
MonoGS: Resetting the system
MonoGS: Initialized map
MonoGS: Evaluating ATE at frame:  23
Eval: RMSE ATE [m] 0.006950586853140792
Traceback (most recent call last):
  File "slam.py", line 252, in <module>
    slam = SLAM(config, save_dir=save_dir)
  File "slam.py", line 110, in __init__
    self.frontend.run()
  File "/root/MonoGS/utils/slam_frontend.py", line 473, in run
    monocular=self.monocular,
  File "/root/MonoGS/utils/eval_utils.py", line 110, in eval_ate
    monocular=monocular,
  File "/root/MonoGS/utils/eval_utils.py", line 50, in evaluate_evo
    fig = plt.figure()
  File "/root/miniconda3/envs/MonoGS/lib/python3.7/site-packages/matplotlib/pyplot.py", line 809, in figure
    FigureClass=FigureClass, **kwargs)
  File "/root/miniconda3/envs/MonoGS/lib/python3.7/site-packages/matplotlib/pyplot.py", line 324, in new_figure_manager
    _warn_if_gui_out_of_main_thread()
  File "/root/miniconda3/envs/MonoGS/lib/python3.7/site-packages/matplotlib/pyplot.py", line 314, in _warn_if_gui_out_of_main_thread
    if (_get_required_interactive_framework(_get_backend_mod())
  File "/root/miniconda3/envs/MonoGS/lib/python3.7/site-packages/matplotlib/pyplot.py", line 217, in _get_backend_mod
    switch_backend(dict.__getitem__(rcParams, "backend"))
  File "/root/miniconda3/envs/MonoGS/lib/python3.7/site-packages/matplotlib/pyplot.py", line 293, in switch_backend
    newbackend, required_framework, current_framework))
ImportError: Cannot load backend 'TkAgg' which requires the 'tk' interactive framework, as 'headless' is currently running
MonoGS: Resetting the opacity of non-visible Gaussians

D31T4, 2025-01-10:
Seems Evo changed their API in later versions.

The `align_trajectory` function is removed. Use the class method `PosePath3D.align(...)` instead.

Below is what I changed in `utils/eval_utils.py`.
```{python}
def evaluate_evo(poses_gt, poses_est, plot_dir, label, monocular=False):
    ## Plot
    traj_ref = PosePath3D(poses_se3=poses_gt)
    traj_est_aligned = PosePath3D(poses_se3=poses_est)
    traj_est_aligned.align(
        traj_ref, correct_scale=monocular
    )

    ...
```

MichaelCurrie, 2025-04-04:
I included the fix given by D31T4 at #183 so this can be closed.