muskie82/MonoGS #167: evo.core.geometry.GeometryException: Degenerate covariance rank, Umeyama alignment is not possible

Wingerllyyy, 2024-12-07:
Thank you for your great work!
I got the following error when running the example code.  
 python slam.py --config configs/mono/tum/fr3_office.yaml_

I am using Ubuntu 22.04, cuda 12.4, and installed cudatoolkit 11.6.2, evo 1.11.0, torchvision 0.13.1 in the conda environment
[image]

Wingerllyyy, 2024-12-09:
发现是更改了代码导致，重新下载源码后正常

closed 2024-12-09 by Wingerllyyy

Michelvl92, 2024-12-19:
What have you changed? I have the same problem, but I'm not sure how to fix it. When I disable evaluation then this is not happening. I use my own TUM style mono and RGB-D dataset.

Wingerllyyy, 2025-02-22:
Sorry for late reply! 

I think I have changed the code 'MONOGS/utils/dataset.py' in class TUMParser, when I use the author's code, it goes well.

Regards,

Yangwen

Message ID: ***@***.***>