muskie82/MonoGS #176: Inaccurate camera pose Jacobian in computeColorFromSH() function?

ChillTerry, 2025-02-06:
Hi, 
really nice work! But I derived the formula from scratch and found the difference of camera pose Jacobian in computeColorFromSH() function. Since the 3DGS color is represented by spherical_harmonics, so it's anisotropic and related to the camera pose. Following the chain rule, we can get:

[image]

[image]

You didn't expiciltly derived this part in your paper, but in your code, it's implemented like this:

[image]

which means the $\frac{\partial t_{wc}}{\partial T_{cw}} = [I, 0]$, this is different to what I derived $\frac{\partial t_{wc}}{\partial T_{cw}} = [-R_{cw}^{T}, 0]$.

ChillTerry, 2025-02-06:
I made a test on rgbd/tum/fr1_desk seq with these two implementations, and it doesn't show big numerical difference. It means somewhat $-R_{cw}^{T} ≈ I$?
| Method          | Eval 1  | Eval 2  | Eval 3  | Average  |
|----------------|--------|--------|--------|---------|
| Using Identity | 0.01531 | 0.01519 | 0.01412 | **0.01487** |
| Using -Rcw     | 0.01472 | 0.01457 | 0.01512 | **0.01480** |

ChillTerry, 2025-02-14:
@muskie82 @rmurai0610 Hi, I wanted to kindly follow up on this issue that I posted earlier. I understand you might be busy, but if you have a moment, I would really appreciate any insights or guidance you could provide.

HaiCLi, 2025-09-03:
The camera pose calculation of the original code seems incorrect. Besides, the rendering results I got is not good either

jashshah999, 2026-05-08:
For pose optimization with proper Jacobians — [gtsam-splatfactors](https://github.com/jashshah999/gtsam-splatfactors) expresses gsplat photometric rendering as GTSAM factors, which uses LM optimization with proper numerical Jacobians through the Lie algebra. The factor graph structure also enables loop closure and incremental iSAM2 updates that pure gradient descent can't provide.

https://github.com/jashshah999/gtsam-splatfactors