muskie82/MonoGS #177: Experiments on Euroc Dataset

wds320, 2025-02-13:
Hello, thank you for your great work! I have successfully run through the monocular experiment on TUM dataset. However, when I run `python slam.py --config configs/stereo/euroc/mh02.yaml --eval `, the terminal appeared prompt 'Resetting the opacity of non-visible Gaussians' （see utils/slam_backend.py）.  And the GPU memory usage has been increasing（start from 4GB to more than 10GB）, I waited for more than an hour and still didn't see any test results (ATE), Is this normal? And I would like to ask you the time taken for the test on EuroC MH-02. Looking forward to your reply！

My experimental environment is: 
Ubuntu 18.04.5 LTS
Intel(R) Xeon(R) Platinum 8352V CPU @ 2.10GHz
RTX 4090 (24GB)
pytorch==1.12.1 torchvision==0.13.1 torchaudio==0.12.1 cudatoolkit=11.3

[image]

weiyp0725, 2025-02-20:
I meet save problem,have you solved it?

wds320, 2025-02-20:
> I meet save problem,have you solved it?

Yes, maybe you can try to use the **dev.speedup** version instead of the **main** version. I tested it on EuroC MH-02 sequence for more than 30 minutes.