muskie82/MonoGS #179: How did you get the depth_scale value?

Tigro-Amur, 2025-03-10:
Hi, thank you for sharing this great code. I would like to run MonoGS on my custom dataset.

In the configuration file located at /configs/rgbd/replica/base_config.yaml, there is a parameter called depth_scale that needs to be specified, as shown in the image below.

[image]

How did you determine this depth_scale value? I checked my camera’s product brochure, but I couldn’t find any information about it. If you used a specific method or code to obtain this value, I would really appreciate it if you could share it.

Thank you in advance!

lemonci, 2025-03-12:
Hello, I think this can help if your data is recorded with RealSense:

https://github.com/IntelRealSense/librealsense/issues/3473

I would like to run it on my own dataset as well. May I ask the directory structure of your dataset? Thanks.

Tigro-Amur, 2025-03-13:
Thank you for answer but I used ImageSource camera. So I can't use those code unfortunately. I think I need more general way to get a depth scale from acquired depth images.
I structured my dataset's directory the same way as the Replica dataset.