muskie82/MonoGS #181: Why Does Monocular Mode Require Depth Files?

betaandmartin, 2025-03-13:
Hi dear Author's 

I noticed that when running in monocular mode, the dataset still requires depth files. I was expecting that depth data wouldn't be needed for monocular setups. Could you clarify why depth files are still required?
Is depth data used for any preprocessing, initialization, or evaluation steps even in monocular mode?
Would it be possible to run the system without depth files, and if not, what part of the code depends on them?
I’d really appreciate your insights on this. Thanks for your time!
Best regards,

yangxu-debug, 2025-07-03:
the same question!

1425562478, 2025-07-03:
same question！

jashshah999, 2026-05-08:
gtsam-splatfactors handles monocular initialization differently — it uses Depth Anything V2 as a one-shot init, then refines scale jointly with poses via photometric factors in a GTSAM graph. No depth files needed at runtime. Also has iSAM2 loop closure which MonoGS lacks.

https://github.com/jashshah999/gtsam-splatfactors